JamesCJ60/Universal-x86-Tuning-Utility #2: Clarification about functionality

philipjan, 2022-08-05:
Since i keep enabling **Performance Mode** and **undervolt**. Just want to clarify, do I need to **select/click** them everytime the app starts?

[image]

[image]

philipjan, 2022-08-05:
Because as i check on the preset txt file, there's no persisted value for them.

philipjan, 2022-08-07:
@JamesCJ60

JamesCJ60, 2022-08-07:
From memory, the power profiles wouldn't need reapplying on boot if your using auto reapply. Curve optimiser needs to be reapplied every boot as a safety measure.

philipjan, 2022-08-11:
Thank you so much for the response. Appreciated. @JamesCJ60

closed 2022-08-18 by JamesCJ60