JamesCJ60/Universal-x86-Tuning-Utility #7: Not starting at system boot and settings not keeping checked

mateus558, 2022-09-01:
First of all this is an awesome tool, it works very well and I'm getting better results than the ryzen controller software.

But there are two problems that I found a bit anoying.  The first is that it doesn't start at boot on win 11, the second one is that if I check the CPU option on "Clock control" tab and change to other tab and go back it is disselected, so I'm not sure if it is enabled and if I need to reapply it.
It also would be nice if it had an option to apply a profile when the system boots.

NostromoUS, 2022-09-09:
I'd like to echo this issue.

1. - Start at boot not working at all.
2. - When applying anything (I'm only undervolting) if you move away from that tab and back the checkbox becomes unticked, which can cause confusion even though it's applied.

JamesCJ60, 2022-09-30:
Hi, this should now be fixed in the latest beta.

closed 2022-09-30 by JamesCJ60