JamesCJ60/Universal-x86-Tuning-Utility #10: 6800HS Support

NalCol, 2022-09-10:
It will crash my Windows everytime I change performance preset, is 6800HS in support?

I saw this program installed inpoutx64.sys, can I clean uninstall this program?

System: Windows 11 Beta 22622.590
Computer: Yoga Slim 7 Pro 14ARH7/XiaoXinPro 14 ARH5R/Slim 7 14ARH

JamesCJ60, 2022-09-10:
Hi, 
it just makes sure that drivers that should be part of a normal Windows install are actually installed.

6900HS is supported by UXTU, that menu still uses RyzenAdj for the time being, so there’s probably a compatibility issue with the latest Windows 11 beta.

closed 2022-09-10 by JamesCJ60

NalCol, 2022-09-11:
> Hi, it just makes sure that drivers that should be part of a normal Windows install are actually installed.
> 
> 6900HS is supported by UXTU, that menu still uses RyzenAdj for the time being, so there’s probably a compatibility issue with the latest Windows 11 beta.

Thanks!