JamesCJ60/Universal-x86-Tuning-Utility #14: CPU frequency tweak base on cores active.

Godl1nk, 2022-10-14:
Suggestion: adding a feature which allows us to set cpu frequency according to number of cores active, smtg like CPU turbo ratio limit on throttlestop

JamesCJ60, 2022-11-30:
I can add something along the lines of this in the future, but it would be limited to the following:
- Unlocked CPUs/APUs
- Systems that support Windows power plan max frequency setting

Godl1nk, 2022-12-07:
yes im requesting just that "Systems that support Windows power plan max frequency setting"

i request for a feature like ThrottleStop's clock ratio which is only for intel to be added fro amd in this tool, allowing us to set cpu max frequency based on cores active, which u can limi the frequency with windows power plan, just need to add a listener to check number of cores under load