JamesCJ60/Universal-x86-Tuning-Utility #19: 6800hs settings don't seem to be applying

blakemoobs, 2022-12-01:
I'm wondering if there are nuances to using this along with armoury crate - I find that the undervolt or AP mode never seem to apply and armoury crate just does what it wants. Testing with cinebench I find that my 6800hs just hits 95 degrees within the first 10 seconds and clock speed drops pretty sharply. Is it likely that I have a bad paste from factory (zephyrus g15 2022 with liquid metal apparently)? I find that using turbo or manual mode I'm idling at 60 to 70 degrees. 

I had one instance where I felt like the undervolt applied and I had a sustained 4.0xx ghz cinebench run and my score went from 11500 to 13500 - otherwise the thermals don't allow it to go above 3.6-3.8 and it just hovers around 95degrees.

Would love some help navigating this - or is the 6800hs just not undervolt-able?

littlebloke, 2022-12-08:
I have a ROG Flow X16 with the same Processor and it looks like i'm having the same issue. and same question as you. You may want to try a previous build, I think 7.1 worked. I will veryify at some point. I believe that 6800HS can be undervolted.

blakemoobs, 2022-12-08:
Ahhh OK, thanks for testing for me! I've read around and it seems like the 6900hs can be under volted but not the 6800 - there was the one miraculous time it worked and all my thermals dropped and everything was great but I have no idea how to replicate that. I'm just not sure if I'm supposed to do things in a particular order or if there is a known issue. I have had the aatu app crash my laptop a few times which sortve indicates it was doing something but 99% nothing happens at all. I've had it for a week now and am already tired of hearing max fans and watching fps drop due to thermals.

littlebloke, 2022-12-08:
This was the post that got me onto undervolting, its a bit old now but there may be some information there related to the 6800? *shrug*:

https://www.reddit.com/r/ZephyrusG15/comments/v0jc03/how_to_undervolt_your_cpu_5900hs6900hs/

I definitely did have some luck before, and my system would crash around the -20/24 mark, so something was happening before I usually left it sitting at -16, which seemed more or less stable... and it did have an impact on temps... now though i'm not seeing any appreciable difference... and no crashing!

littlebloke, 2022-12-10:
Just to follow up, it looks as though the issue is that it does not retain the settings after sleeping. If I put the computer to sleep and wake, then run cinebench r23 or some other CPU intensive task, the temp goes up to max (i.e. 95), if I open UXTU and re-apply the temps are reduced back to the levels set in the software (i.e. 80) could someone verify?

closed 2022-12-18 by JamesCJ60

blakemoobs, 2022-12-18:
Hey @JamesCJ60 did you end up verifying if it works with the 6800hs? I appreciate your work!