JamesCJ60/Universal-x86-Tuning-Utility #27: App based preset selection [FEATURE REQUEST]

l0lGhostlxl, 2023-01-05:
**Is your feature request related to a problem? Please describe.**
When trying to use the AATU different applications have different power requirements, therefore you must switch between different presets, and this takes longer to switch between presets manually especially if the laptop is being taxed due to an application that is not suited to the current preset.  Therefore, automatically switching between this would be best.

**Describe the solution you'd like**
Depending on what application is currently open or in focus the preset can automatically switch. For example if App A is launched the system will use the "Standard" preset, and another App have another preset. 

A possible issue is if there are multiple apps with different presets it might keep switching between presets so the best solution would be to use the preset of the app the user currently has focus on.

**Describe alternatives you've considered**
Manually switching

panther7, 2023-01-23:
+1 for presets like in AATU

JamesCJ60, 2023-06-27:
Hi, this is possible using the adaptive mode within UXTU V2.

closed 2023-06-27 by JamesCJ60