JamesCJ60/Universal-x86-Tuning-Utility #39: [BUG] Riot Vanguard Warning

zanz9, 2023-02-09:
After Install Driver inpoutx64 Vanguard cursing 

Vanguard has blocked the following file from loading on your system:

"\Device\HarddiskVolume3\Windows\System32\drivers\inpoutx64.sys"

To disable Vanguard, please right-click on the Vanguard notification icon and select "Exit Vanguard".

NOTE: This will put your system in an untrusted state. You will have to restart in order to play games that depend on Vanguard's functionality.

Would you like to learn more?

[image]

 - Windows 10 22H2

zerkzis13, 2023-02-13:
Hello I want to play valorant again, and I don't know how to uninstall your program? Please can you help me?

zerkzis13, 2023-02-14:
Hello please help figure out how to remove your program from my pc. I just want to play valorant. thank you!

JamesCJ60, 2023-03-06:
> After Install Driver inpoutx64 Vanguard cursing
>
> Vanguard has blocked the following file from loading on your system:
>
> "\Device\HarddiskVolume3\Windows\System32\drivers\inpoutx64.sys"
>
> To disable [...]

To bypass the blocking system of Riot Vanguard when inpout64 is detected, enable Core Isolation in Windows Security. 

This method requires a restart and will still display a message from Riot Vanguard that inpout64 has been blocked even though it hasn't and both UXTU and Vanguard will run without issue.

closed 2023-03-06 by JamesCJ60