JamesCJ60/Universal-x86-Tuning-Utility #41: [BUG] resolution is incorrect on IdeaPad Slim 5

Mattwmaster58, 2023-02-09:
**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to project snowdrop

**Expected behaviour**
Correct display resolution

**Desktop (please complete the following information):**
 - OS Version: Windows 10 21H2

l0lGhostlxl, 2023-02-09:
I also do not get the correct resolution in the application as well 
[image]

JamesCJ60, 2023-04-28:
I’m currently clearing all UXTU V1.0.0 issues now that UXTU V2.0.0 is in beta testing. If the issue still persists in UXTU V2.0.0 feel free to reopen the issue.

closed 2023-04-28 by JamesCJ60