JamesCJ60/Universal-x86-Tuning-Utility #42: Linux support

fabifont, 2023-02-10:
Will you ever support Linux?

AvicennaJr, 2023-02-14:
Came here for the same question ^

ankitpaudel20, 2023-02-19:
one up.

JamesCJ60, 2023-02-22:
Hi, the current application GUI has no plans to support Linux anytime soon. Though in the future, a dedicated UXTU GUI will be created for Linux.

closed 2023-02-22 by JamesCJ60

Hazmi35, 2023-08-13:
Wouldn't it make sense for this issue to be kept open? As a way to track, people can susbcribe to this issue and if it's implemented then they would also know.

I'm looking forward for it

Okazakee, 2023-08-25:
> Wouldn't it make sense for this issue to be kept open? As a way to track, people can susbcribe to this issue and if it's implemented then they would also know.
> 
> I'm looking forward for it

Yes, it doesn't bother imho to have it open, just a reminder.

KyunLFA, 2023-09-19:
Hi, would love to see it

FedFer98123, 2023-10-30:
This should be open if it is a planned feature

bogn83, 2023-11-01:
For the time being there is still https://gitlab.com/ryzen-controller-team/ryzen-controller#debian-like

Maybe Linux people willing to help should comment into this issue here with the way they could support. I can program JavaScript and thus Electron though I don't have experience with the latter. Mentioned Electron because the above ryzen controller is written in it and had people maintaining linux support for it.

Would also be willing to dive into Rust and use any of the tools from this reddit post
> klorophane 3 days ago
> Egui is considered by many to be one of the simplest GUIs there is in Rust. Lots of people love it.
> 
> Aside from that:
> 
> [iced](https://github.com/iced-rs/iced) has a lot of features and is supported by Cosmic DE and PopOs. I have used it and found it quite pleasing to use, although the elm-like model is not for everyone.
> 
> [tauri](https://tauri.app/) is the easiest way to get started if you know a web framework. Write the frontend in whatever web ui framework, and the backend in Rust. I worked with Rust + Svelte and I really liked it a lot!
> 
> [slint](https://slint.dev/) Has a fully-featured custom markup language that seamlessly integrates with Rust (and other languages). It was created by some influential people that worked on Qt. I really like it, but it wasn't mature enough for complex desktop applications last time I checked (that was a long time ago, granted). I believe you need a license for non open-source projects, but it's worth considering.

cybik, 2024-01-03:
At this point in time, RyzenCtl is considered deprecated. Current Ubuntu packages do not run on latest (Mantic), and they *themselves* are saying
* that the project is deprecated
* to use this very utility

[image]