JamesCJ60/Universal-x86-Tuning-Utility #45: [BUG] Wndows 10 Applciation crash

RazielsOwn, 2023-03-06:
**Describe the bug**
Windows 10 Pro Application crash.

**To Reproduce**
Steps to reproduce the behaviour:
Happens sometime...

Event Viewer logs shows 2 related errors:
1) Application error
```
Faulting application name: Universal x86 Tuning Utility.exe, version: 1.0.0.5, time stamp: 0xa41389a4
Faulting module name: KERNELBASE.dll, version: 10.0.19041.2546, time stamp: 0xe8e9ac9b
Exception code: 0xe0434352
Fault offset: 0x000000000002cd29
Faulting process id: 0x3374
Faulting application start time: 0x01d94ffc947764a0
Faulting application path: D:\Install\UXTU\Universal x86 Tuning Utility.exe
Faulting module path: C:\Windows\System32\KERNELBASE.dll
Report Id: 29b7f3e3-1833-49be-8088-15e0ccef373d
```

2) .NET Runtime error
```
Application: Universal x86 Tuning Utility.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.ApplicationException
   at System.Threading.Mutex.ReleaseMutex()
   at RyzenSmu.Smu.SendMsg(UInt32, UInt32, UInt32, UInt32, UInt32[] ByRef)
   at AATUV3.Pages.Info.<updateInfoCZN>b__7_0()
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at MS.Internal.CulturePreservingExecutionContext.Run(MS.Internal.CulturePreservingExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndWrapper.WndProc(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(System.Windows.Threading.DispatcherPriority, System.TimeSpan, System.Delegate, System.Object, Int32)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr, Int32, IntPtr, IntPtr)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(System.Windows.Interop.MSG ByRef)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Application.RunDispatcher(System.Object)
   at System.Windows.Application.RunInternal(System.Windows.Window)
   at AATUV3.App.Main()
```

**Desktop (please complete the following information):**
 - OS Version: Windows 10 Pro 22H2 
 - APP version 1.1.0 Beta 3

JamesCJ60, 2023-04-28:
I’m currently clearing all UXTU V1.0.0 issues now that UXTU V2.0.0 is in beta testing. If the issue still persists in UXTU V2.0.0 feel free to reopen the issue.

closed 2023-04-28 by JamesCJ60

Jackpot254, 2024-10-14:
Reopening because this is currently happening on a new installation of X86 Tuning. KERNELBASE.dll fails, with an additional .NET runtime issue, "Could not load file or assembly 'PresentationFramework, Version=8.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'"