JamesCJ60/Universal-x86-Tuning-Utility #48: [BUG] System uses 8-10% of CPU when UXTU is running

chorong761, 2023-03-19:
**Describe the bug**
When I launch UXTU, the system process would constantly use 8-10% of my CPU (5900HS) and holding it in boost clocks.

**To Reproduce**
Steps to reproduce the behaviour:
1. Launch UXTU
2. Check Task Manager
3. Sees the high CPU usage for "System"

**Laptop (please complete the following information):**
Zephyrus G14 2021, Windows 11 22H2, UXTU v1.0.3 Stable

nsilvias, 2023-03-20:
1.1.0 beta 4 and 5 gave me high cpu usage from system and controller app as well. downgrading to 1.1.0 beta 3 fixed this issue for me.

JamesCJ60, 2023-04-28:
I’m currently clearing all UXTU V1.0.0 issues now that UXTU V2.0.0 is in beta testing. If the issue still persists in UXTU V2.0.0 feel free to reopen the issue.

closed 2023-04-28 by JamesCJ60