JamesCJ60/Universal-x86-Tuning-Utility #49: [BUG] Sensors stopped working

killerart, 2023-03-22:
The info tab stopped showing any info. All the sensors values are 0.

It worked before, but one day it stopped working. Didn't really change anything about my system.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS Version: Windows 10 22H2 build 19045.2728
 - CPU: AMD Ryzen 5 5900HX

marwan4mr, 2023-03-25:
i have  the same issue with 5800h lenovo legion

stujii, 2023-03-29:
+1 (lenovo legion 5800h)

JamesCJ60, 2023-04-28:
I’m currently clearing all UXTU V1.0.0 issues now that UXTU V2.0.0 is in beta testing. If the issue still persists in UXTU V2.0.0 feel free to reopen the issue.

closed 2023-04-28 by JamesCJ60

killerart, 2023-04-29:
@JamesCJ60 well, the issue fixed itself without me doing anything. I didn’t update windows, bios or the app. Started the laptop one day and it worked.