JamesCJ60/Universal-x86-Tuning-Utility #50: [BUG] App crashes instantly on launch

0x8008, 2023-04-02:
**Describe the bug**
Opening the app leads to instant crashes 99.9% of the time. It sometimes opens but it is a very rare occurance.

**To Reproduce**
Steps to reproduce the behaviour:
1. Double-click on `Universal x86 Tuning Utility.exe`
2. Witness it crash immediately in task manager

**Expected behaviour**
The app to launch.

**Desktop (please complete the following information):**
 - OS Version: Win11 x64 22H2 22621.1413

**Additional context**
Willing to provide any neccessary logs etc. if you have something in mind that I might need to attach.

raenye, 2023-04-05:
I have something similar, still not solved.
Look at the Event Viewer->Windows Logs->Application.
Mine shows this:
```
Application: Universal x86 Tuning Utility.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.Xml.XmlException
   at System.Xml.XmlTextReaderImpl.Throw(System.Exception)
   at System.Xml.XmlTextReaderImpl.ParseText(Int32 ByRef, Int32 ByRef, Int32 ByRef)
   at System.Xml.XmlTextReaderImpl.ParseText()
   at System.Xml.XmlTextReaderImpl.ParseElementContent()
   at System.Xml.XmlTextReaderImpl.Skip()
   at System.Configuration.XmlUtil.StrictSkipToNextElement(System.Configuration.ExceptionAction)
   at System.Configuration.BaseConfigurationRecord.ScanSectionsRecursive(System.Configuration.XmlUtil, System.String, Boolean, System.String, System.Configuration.OverrideModeSetting, Boolean)
   at System.Configuration.BaseConfigurationRecord.ScanSectionsRecursive(System.Configuration.XmlUtil, System.String, Boolean, System.String, System.Configuration.OverrideModeSetting, Boolean)
   at System.Configuration.BaseConfigurationRecord.ScanSections(System.Configuration.XmlUtil)
   at System.Configuration.BaseConfigurationRecord.InitConfigFromFile()

Exception Info: System.Configuration.ConfigurationErrorsException
   at System.Configuration.ConfigurationSchemaErrors.ThrowIfErrors(Boolean)
   at System.Configuration.BaseConfigurationRecord.ThrowIfParseErrors(System.Configuration.ConfigurationSchemaErrors)
   at System.Configuration.ClientConfigurationSystem.OnConfigRemoved(System.Object, System.Configuration.Internal.InternalConfigEventArgs)

Exception Info: System.Configuration.ConfigurationErrorsException
   at System.Configuration.ClientConfigurationSystem.OnConfigRemoved(System.Object, System.Configuration.Internal.InternalConfigEventArgs)
   at System.Configuration.Internal.InternalConfigRoot.OnConfigRemoved(System.Configuration.Internal.InternalConfigEventArgs)
   at System.Configuration.Internal.InternalConfigRoot.RemoveConfigImpl(System.String, System.Configuration.BaseConfigurationRecord)
   at System.Configuration.BaseConfigurationRecord.GetSectionRecursive(System.String, Boolean, Boolean, Boolean, Boolean, System.Object ByRef, System.Object ByRef)
   at System.Configuration.BaseConfigurationRecord.GetSection(System.String)
   at System.Configuration.ConfigurationManager.GetSection(System.String)
   at System.Configuration.ClientSettingsStore.ReadSettings(System.String, Boolean)
   at System.Configuration.LocalFileSettingsProvider.GetPropertyValues(System.Configuration.SettingsContext, System.Configuration.SettingsPropertyCollection)
   at System.Configuration.SettingsBase.GetPropertiesFromProvider(System.Configuration.SettingsProvider)
   at System.Configuration.SettingsBase.GetPropertyValueByName(System.String)
   at System.Configuration.SettingsBase.get_Item(System.String)
   at System.Configuration.ApplicationSettingsBase.GetPropertyValue(System.String)
   at System.Configuration.ApplicationSettingsBase.get_Item(System.String)
   at AATUV3.MainWindow..ctor()

Exception Info: System.Windows.Markup.XamlParseException
   at System.Windows.Markup.XamlReader.RewrapException(System.Exception, System.Xaml.IXamlLineInfo, System.Uri)
   at System.Windows.Markup.WpfXamlLoader.Load(System.Xaml.XamlReader, System.Xaml.IXamlObjectWriterFactory, Boolean, System.Object, System.Xaml.XamlObjectWriterSettings, System.Uri)
   at System.Windows.Markup.WpfXamlLoader.LoadBaml(System.Xaml.XamlReader, Boolean, System.Object, System.Xaml.Permissions.XamlAccessLevel, System.Uri)
   at System.Windows.Markup.XamlReader.LoadBaml(System.IO.Stream, System.Windows.Markup.ParserContext, System.Object, Boolean)
   at System.Windows.Application.LoadBamlStreamWithSyncInfo(System.IO.Stream, System.Windows.Markup.ParserContext)
   at System.Windows.Application.DoStartup()
   at System.Windows.Application.<.ctor>b__1_0(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at MS.Internal.CulturePreservingExecutionContext.Run(MS.Internal.CulturePreservingExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndWrapper.WndProc(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
[... cut ...]
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr, Int32, IntPtr, IntPtr)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(System.Windows.Interop.MSG ByRef)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Application.RunDispatcher(System.Object)
   at System.Windows.Application.RunInternal(System.Windows.Window)
   at AATUV3.App.Main()
```

(if yours is a different problem, let me know and I'll open a new issue to avoid hijacking this one)

DarkSpectre066, 2023-04-11:
I was having a similar issue with UXTU immediately crashing after startup and found my solution on Discord:
https://discord.com/channels/772105072720871435/1086777866232016897/1086804561722875934
"Delete the AATUV3 folder found in either AppData/Local or AppData/LocalLow" -JamesCJ

JamesCJ60, 2023-04-28:
I’m currently clearing all UXTU V1.0.0 issues now that UXTU V2.0.0 is in beta testing. If the issue still persists in UXTU V2.0.0 feel free to reopen the issue.

closed 2023-04-28 by JamesCJ60