JamesCJ60/Universal-x86-Tuning-Utility #55: [BUG]curve optimiser settings not support 7735h

philipzha, 2023-04-19:
**Describe the bug**
A clear and concise description of what the bug is that i change curve optimiser settings -all core to -30 even to -99,but it doesn't work.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS Version: [e.g. Windows 11 22H2]

**Additional context**
Add any other context about the problem here.

JamesCJ60, 2023-04-19:
Hi, this option only works on Ryzen 9’s are the 4000 series.

closed 2023-04-19 by JamesCJ60

philipzha, 2023-04-19:
thanks,that' a sad story