JamesCJ60/Universal-x86-Tuning-Utility #67: Closes after opening the premade section [BUG]

rugo-cafe, 2023-05-14:
If i click in the premade presets section either by the sidebar icon, or the button below the welcome message, it closes itself instantly.

V2.0.0 Beta Build 3 
OS Version: Win 11 22H2
CPU: Ryzen 5 3500U

JosivaldoLira123, 2023-05-15:
I was using Win 11 22H1 and there was no problem with the app. Yesterday I formatted my PC and installed Win 11 22H2, now I have the same problem as you. Seems to be a bug with this version of Windows.

JosivaldoLira123, 2023-05-15:
If you already used the app in this version before, it could be a bug with the processor or even the driver, my processor is the R5 5500u and the driver version is 23.4.3. I hope the developer fixes this problem soon.

JamesCJ60, 2023-05-16:
Hi, this should be fixed as of the latest build.

MentolosNokedli, 2023-05-17:
Hi, on my Lenovo 82jy Legion 5 17ach6h (CPU 5800H) whit Win 11 full updated only the  V1.0.3 Stable version is working.
All other version crashes @ Premade and  in Custome iff i press the Apply current settings or Save Buttion. Sorry for bad English

MortiMaus, 2023-05-21:
Hi, have the same issue with closing the application after clicking the premade button. 
Im on an Asus TUF A17 with 6800h. 
Im on 2.0.0.8 my os is Win11pro 22H2
Any help is welcome. :-)

jwagra, 2023-06-12:
Hi, i have the same issue on [V2.0.0 - Beta 6.1] on my Acer Nitro 5 - Ryzen 6600H with latest win11pro

JamesCJ60, 2023-06-25:
Hi, this issue should now be fixed for everyone as of the latest beta.

closed 2023-06-25 by JamesCJ60

YegorStepanov, 2023-06-25:
beta7 is still crashing