JamesCJ60/Universal-x86-Tuning-Utility #68: [BUG] Skin Temperature Limit (apu-skin-temp) value cannot be set properly

Snuupy, 2023-05-14:
**Describe the bug**

Skin Temperature Limit (`apu-skin-temp`) is set to a nonsense value

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to custom presets
2. Set skin temp limit to something reasonable, such as 60C (or manually via `ryzenadj --apu-skin-temp 60`
3. Apply, run `ryzenadj --info`
4. See: `| STT LIMIT APU       |    0.407 | apu-skin-temp      |`

**Expected behaviour**

apu-skin-temp should be set to 60.

**Screenshots**

N/A

**Desktop (please complete the following information):**
 - OS Version: Windows 11 22H2

JamesCJ60, 2023-06-25:
Hi, I have not been able to reproduce this issue on my end, so this is possibly just a reporting issue.

closed 2023-06-25 by JamesCJ60

Snuupy, 2023-06-28:
Not a reporting issue, but it's because the ThinkPad T14 AMD BIOS keeps reverting the TDP changes back. I have had to run this instead in a loop: `while (1) {C:\path\ryzenadj-win64\ryzenadj -a 25000 -b 25000 -c 25000 --apu-skin-temp 60 --tctl-temp=86; sleep 60}`

JamesCJ60, 2023-06-28:
Auto reapply in the settings menu exists for this reason.

Snuupy, 2023-12-08:
> Auto reapply in the settings menu exists for this reason.

You are correct, I did not grok/understand the flags to a sufficient degree (and also the BIOS resetting the limits every "once in a while" is very confusing/opaque/hard to debug

Thank you for all the releases and allowing me to set my laptop's TDPs! It's a shame Lenovo sets such low limits by default.