JamesCJ60/Universal-x86-Tuning-Utility #69: Adaptive performance when toggled on shows a dll error[BUG]

l0lGhostlxl, 2023-05-15:
**Describe the bug**
When I toggle it on it shows a dll error
**To Reproduce**
Steps to reproduce the behaviour:
1. Toggle it on

**Expected behaviour**
No errors

**Screenshots**
[image]

**Desktop (please complete the following information):**
Edition	Windows 11 Home
Version	22H2
Installed on	‎20/‎03/‎2023
OS build	22621.1702
Experience	Windows Feature Experience Pack 1000.22641.1000.0

l0lGhostlxl, 2023-05-15:
It also causes my laptop to crash after a bit as well. The laptop is a flow x16 with a Ryzen 9 6900HS 2022 edition

siloli, 2023-05-15:
Hi,
Same error here with the flow x13 Ryzen 9 5900HS.
Windows 11 Pro Build 22621

JamesCJ60, 2023-05-16:
Hi, this should be fixed as of the latest build.

l0lGhostlxl, 2023-05-17:
The new build just seems to crash immediately after launching.

l0lGhostlxl, 2023-05-17:
Never mind a quick restart has fixed it, and the app and feature is working now. Thank you for your help.

closed 2023-05-17 by l0lGhostlxl