JamesCJ60/Universal-x86-Tuning-Utility #71: [BUG]

dutch45612, 2023-05-17:
**Describe the bug**
System crash/ freeze entirely when starting adaptive mode(need to force close the laptop holding down the power button).

**To Reproduce**
Start adaptive mode with a minimum GPU to any value, it happens randomly within about 2 minutes.

**Expected behaviour**
Automatically managing TDP/ Gpu-CPU clock frequency according to load.

**Screenshots**
Black screen, the sound are still working but everything else(including inputs) are frozen.

**Desktop (please complete the following information):**
 - OS Version: Windows 11 22H2

**Additional context**
- My device : GPD WIN MAX 2
- AMD Ryzen 7 6800U with Radeon Graphics (680m)
- RAM 32.0 GB

igornovitzky, 2023-05-17:
I get this too on 6850u lenovo z13 32gb RAM and Radeon graphics, latest windows 11

Tygr09, 2023-05-19:
I got that as well.  Lenovo amd ryzen 7 6800 with 3070 graphics and windows 11.  My laptop crashed within 2 minutes.

filminghub, 2023-05-29:
> I got that as well. Lenovo amd ryzen 7 6800 with 3070 graphics and windows 11. My laptop crashed within 2 minutes.

I have the same specs except I'm using Asus laptop. And I got crashed by turning on performance mode. Do you still have the same issue?

igornovitzky, 2023-05-30:
My 6850u Lenovo Z13 AMD crashes after 3-4min of performance mode
Igor

JamesCJ60, 2023-06-26:
Hi, this should no longer be an issue as of the latest betas.

closed 2023-06-26 by JamesCJ60