JamesCJ60/Universal-x86-Tuning-Utility #72: [BUG] [REOPENED] APU Spesific Presets are stopping working while gaming

LatifSC, 2023-05-18:
**Describe the bug**
APU Spesific presets are only working on desktop. While gaming it goes back to default TDP in 1-2 minutes.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Premade'
2. Click on 'Performance or Extreme'
4. See error

**Expected behaviour**
Stay at the selected settings. Also, while in game when I clicked on "Premade" and my PC crashed.

**Desktop (please complete the following information):**
 - OS Version: Windows 10 22H2 (Switched from 11, Clean Install)
Ryzen 7 3750H
2x8GB RAM
GTX 1660Ti

Also, I tried RyzenADJ. The same thing happens with it aswell. Constantly says unexpected value change or something.

LatifSC, 2023-05-18:
#52

female-nectar, 2023-05-22:
Same issue here. 
When using Adaptive Mode Wattage caps to 7.0w every 60sec~.
Couldn't find whats causing the issue...
Setting PBO Limits manually works just fine though.
Ryzen 5 5500u, Win10 22H2, HP 15s-eq2x

JamesCJ60, 2023-06-25:
Hi, this issue should now be fixed as of the latest beta. Otherwise, ensure auto reapply is enabled.

closed 2023-06-25 by JamesCJ60