JamesCJ60/Universal-x86-Tuning-Utility #83: [BUG]WinRing0 load error, the application crushes on lanuch

miasakachenmo, 2023-06-22:
**Describe the bug**
Application crushes on lanuch, log:
Application: Universal x86 Tuning Utility.exe CoreCLR Version: 7.0.723.27404 .NET Version: 7.0.7 Description: The process was terminated due to an unhandled exception. Exception Info: System.TypeInitializationException: The type initializer for 'Universal_x86_Tuning_Utility.Scripts.AMD_Backend.SMUCommands' threw an exception. ---> System.ApplicationException: WinRing OLS_DRIVER_NOT_LOADED at RyzenSmu.Smu..ctor(Boolean EnableDebug) in C:\Users\james\Desktop\Universal x86 Tuning Utility\Universal x86 Tuning Utility\Scripts\AMD Backend\RyzenSmu.cs:line 85 at Universal_x86_Tuning_Utility.Scripts.AMD_Backend.SMUCommands..cctor() in C:\Users\james\Desktop\Universal x86 Tuning Utility\Universal x86 Tuning Utility\Scripts\AMD Backend\SMUCommands.cs:line 15 --- End of inner exception stack trace --- at 

**Desktop (please complete the following information):**
 - OS Version:Windows 11 22H2
 - CPU: AMD  4800H

**Additional context**
It seems to be an error when loading wingring0

Gahwa-Gahwa, 2023-06-23:
I had the same issue, it crashes on startup.

JamesCJ60, 2023-06-25:
Hi, this issue should now be fixed as of the latest beta.

closed 2023-06-25 by JamesCJ60