JamesCJ60/Universal-x86-Tuning-Utility #85: [BUG] Changes not working on Ryzen 5500U

RyzenDude, 2023-06-24:
Greetings,

I have a HP with a AMD 5500U and when I apply the presets they do increase wattage. I utilize HWINFO to monitor and R23 as a tester. When I launch R23 and did my stock benchmark. the wattage would be at 25, however when I launch Universal X86 and try any preset and start the benchmark, the wattage stays at 15 and the CPU clocks don't change. Everything is up to date, Windows 11 and the program itself. I downloaded all the files required for the app to function properly. 

Any advice on how to fix this? Thank you!

Take care.

JamesCJ60, 2023-06-25:
Hi, this issue should now be fixed as of the latest beta. Although, if it still continues, there is a chance your OEM has set a hard 15w limit.

closed 2023-06-25 by JamesCJ60

RyzenDude, 2023-06-27:
Thank you for your response. How is that possible? When I run the computer stock on Cinebench it naturally goes to 25W, how come Universal X86 is making the wattage lower?

ashish7adlakha, 2024-04-21:
i also have the same issue, without the UXTU app the package power goes to 28 watts in and when i open the app the package power gets limited to 15 watts.

erreing, 2024-12-22:
same for me (ryzen 5 5625u)

Parag949, 2025-01-06:
I have Ryzen 7 4780u on my hp prime book. I also have the same issue. although just for 5 min randomly my settings in uxtu applies and my package watage increased . But that was just one time . Now whenerver i apply the presets my chip is automatically not going over 15 w . So now i have to use my system without this utility to have 25w performace

erreing, 2025-01-18:
the best solution is to use an old version like aatu 1.0.1