JamesCJ60/Universal-x86-Tuning-Utility #88: [BUG] Installer refuses to install anywhere than on my Z: drive.

acolwill, 2023-07-01:
**Describe the bug**
Installer refuses to install anywhere than on my Z: drive.

**To Reproduce**
Install application, any options.
Application simply installs into the root of the drive with the last drive letter (z)

**Expected behaviour**
Application should install where I ask it to install.

**Screenshots**
[image]
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 Windows 10 22H2 Build 19045.3155

JamesCJ60, 2023-07-01:
Hi, this is a Microsoft issue not a UXTU issue.

closed 2023-07-01 by JamesCJ60

acolwill, 2023-07-01:
Hi, how is this a Microsoft issue?
Having dealt with MS systems for 30+ years, this is the first time I've seen an installer routine ignore user options.

JamesCJ60, 2023-07-01:
The installer used is from the Visual Studio installer package, which is both created and maintained by Microsoft.
https://marketplace.visualstudio.com/items?itemName=VisualStudioClient.MicrosoftVisualStudio2022InstallerProjects

acolwill, 2023-07-01:
When I remove the drive letter mapping for Z:, the installer routine now installs to the root of D: and not within the directory structure specified.
Do you have a list of available command line switches, or a ZIP/7z download?

acolwill, 2023-07-01:
This feels like an issue with the MSI

Installer log for UXTU
[uxtu.log](https://github.com/JamesCJ60/Universal-x86-Tuning-Utility/files/11927693/uxtu.log)

Installer log for QuickSFV (which also uses VS Installer Projects)
[quicksfv.log](https://github.com/JamesCJ60/Universal-x86-Tuning-Utility/files/11927698/quicksfv.log)