JamesCJ60/Universal-x86-Tuning-Utility #96: [BUG] Extra CPU cycles while idle eats the battery

Charliecoop, 2023-07-12:
**Describe the bug**
The program takes up 0.3-0.7% CPU resources (as reported by task manager) at all times. Fully closing the program allows the laptop battery to last longer due to the CPU cycles saved.

**To Reproduce**
Steps to reproduce the behaviour:
1. Run program, apply a premade preset.
2. Minimise the program to system tray.

**Expected behaviour**
I expected the program to take 0% CPU at all times (maybe 0.1% from time to time if it's reapplying settings or something.

**Screenshots**
(https://imgur.com/a/kPRIaaD)

**Desktop (please complete the following information):**
Windows 11 22H2

**Additional context**
Add any other context about the problem here.

Charliecoop, 2023-07-12:
Current workaround for anyone else having this issue. This involves allowing the program to autorun and then close right away so it can apply the profile and not run in the background anymore (so other automations that require the program to run won't work anymore).

--
1. Create a .bat file with this as the contents:
TASKKILL /F /IM "Universal x86 Tuning Utility.exe"

2. Use Windows Scheduler to run this at the highest privilages at logon.

3. Set Universal x86 Tuning Utility to open and apply the settings at boot.
--

JamesCJ60, 2023-07-29:
This should have been minimised as of the latest builds.

closed 2023-07-29 by JamesCJ60