JamesCJ60/Universal-x86-Tuning-Utility #99: [FEATURE REQUEST] I want to add the ability to apply overclocking parameters right from the moment I open the PC or laptop

dotienanhvietnam, 2023-07-19:
I have customized all the parameters, but every time I restart the laptop or open the laptop, those options have not been applied and only apply the parameters available in the software, so I want an option in the auto section. On the fan section, there is an option to apply overclocking parameters right from the start. Thank you for reading and creating this application thank you❤

[image]

JamesCJ60, 2023-07-24:
Hi, such option already exists and is located within the settings menu.

closed 2023-07-24 by JamesCJ60

dotienanhvietnam, 2023-07-24:
I left my default GPU frequency at 800MHz but every time I turn off my
laptop and turn it back on, I use MSI Afterburner to check that the GPU
frequency is still working by default and I have to go into the app
settings and apply the saved settings again for my GPU frequency to work at
the level I leave