JamesCJ60/Universal-x86-Tuning-Utility #100: Hope to support Ryzen 7045HX series notebook CPU

junjian-yuan, 2023-07-24:
The Ryzen 7045hx series requires further control of TDP, temperature, etc. to achieve noise and performance balance due to thermal power consumption ratio.

I tried version 1.03 and it was able to implement these features, but this version was self-booting and the automatic application Settings feature didn't really work.

Therefore, it is hoped that the new 2.0 version will add CPU support, as well as the ability to automatically apply Settings from the start.

thanks!

TeslaWiki, 2023-07-28:
how to use for Ryzen 7945HX?

cidiousx, 2023-08-03:
Can't it's currently unsupported but also no update on when it would be possibly supported again. I kind of expected this tool to work for my 7945HX since earlier versions did but I'm also waiting for the back-end rewrite to be completed.

junjian-yuan, 2023-08-03:
Ok, thanks!

ASURA715, 2023-09-01:
am waiting for my R9 7945HX this is an amazing App

closed 2024-01-22 by junjian-yuan