JamesCJ60/Universal-x86-Tuning-Utility #112: [BUG]

Arc-2023, 2023-08-04:
**Describe the bug**
The custom preset and adaptive mode settings only take effect for a short period of time, about 1 minute, then they stop working.

**To Reproduce**
Steps to reproduce the behaviour:
Set custom preset or adaptive mode
Wait for about 1 minute
The settings stop working

**Expected behaviour**
The custom preset and adaptive mode settings are expected to keep working continuously, instead of stop working after 1 minute.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS Version: [e.g. Windows 11 22H2]

**Additional context**
The Image Scaling effects seem to randomly stop applying. This does not happen with other settings. Restarting the program is the only way to re-enable adaptive mode once it stops working.

Arc-2023, 2023-08-06:
> i guess the Legion Zone have conflicts with this app.

I didn't even install this

JamesCJ60, 2023-09-01:
Hi, for those case you will need to enable auto-reapply in the settings menu.

closed 2023-09-01 by JamesCJ60