JamesCJ60/Universal-x86-Tuning-Utility #116: [BUG] Laptop Freezes Every Time I Go To Premade Presets

Pandaz111, 2023-08-08:
My laptop Freezes every time I Go To Premade Presets. I have to force restart my laptop whenever it freezes because I cannot do anything else.

I have already deleted and reinstalled this tool about 3 times but have had no luck.

I am using the Asus Zephyrus Laptop G15 with a Ryzen 7 6800HS.

I also have G-Helper installed, does it interfere?

Pandaz111, 2023-08-08:
I am on the newest version of Windows 11

qonstellations, 2023-08-31:
im facing the exact same issue on Dell-G15 5525, Ryzen 5 6600H

AndreeaxD, 2023-11-14:
yea me too on a 7 7745hx ( with a lenovo legion pro 5 ) and i dont know what to do really. . .

jorgepsmatos, 2023-11-17:
Same here. Lenovo legion pro 7, Ryzen 9 7945HX, 4080, Windows 11

It's not every time but like 75% of the time

closed 2023-11-25 by JamesCJ60

MetalRex101, 2024-03-18:
@JamesCJ60 Hey! was it actually fixed? Because I'm getting the same problem with the both latest stable 2.2.5 and beta 2.3.0 versions. I have Lenovo Letion pro 5 with Ryzen 7 7745HX.