JamesCJ60/Universal-x86-Tuning-Utility #118: [BUG]In the latest version 2.0, the application can be launched, but crashes immediately

Seeger0216, 2023-08-12:
**Describe the bug**
In the latest version 2.0, the application can be launched, but crashes immediately

**To Reproduce**
Steps to reproduce the behaviour:
1. This problem occurs just by double-clicking the shortcut on the desktop. The software is currently unavailable.

**Desktop (please complete the following information):**
 - Windows 11 22H2
 - R9-7945HX
 - RTX4060 laptop
 - Legion R9000P 2023

**Additional context**
Do I need to provide any logs? Please tell me how to obtain it.

Prurite, 2023-08-12:
Same issue here.
- Version: v2.0.1
- Windows 11 Professional Edition 22H2
- AMD Ryzen 7 5800H with Radeon Graphics            3.20 GHz
- RTX3060 laptop
- Legion R9000P 2021H

luckyguido, 2023-10-17:
Same issue here:
```
Application: Universal x86 Tuning Utility.exe
CoreCLR Version: 7.0.222.60605
.NET Version: 7.0.2
Description: The process was terminated due to an unhandled exception.
Exception Info: System.TypeInitializationException: The type initializer for 'Universal_x86_Tuning_Utility.Scripts.UXTU_Super_Resolution.Controller' threw an exception.
 ---> System.IO.FileNotFoundException: Could not load file or assembly 'Gma.System.MouseKeyHook, Version=5.7.1.0, Culture=neutral, PublicKeyToken=null'. Impossibile trovare il file specificato.
File name: 'Gma.System.MouseKeyHook, Version=5.7.1.0, Culture=neutral, PublicKeyToken=null'
   at Universal_x86_Tuning_Utility.Scripts.UXTU_Super_Resolution.Controller..cctor()
   --- End of inner exception stack trace ---
   at Universal_x86_Tuning_Utility.Scripts.UXTU_Super_Resolution.Controller.SetUpMagWindow(MainWindow main)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   at System.Windows.Application.RunDispatcher(Object ignore)
   at System.Windows.Application.RunInternal(Window window)
   at Universal_x86_Tuning_Utility.App.Main()
```

seems missing Gma.System.MouseKeyHook

**why are you logging all key pressed while on background??**

closed 2023-11-25 by JamesCJ60

gpz1100, 2024-03-05:
@Seeger0216 I don't know if you ever got this resolved, but I had the same issues on a fresh windows 11 23h2 install.  Anything below v2.1.0 would work, 2.1.0 and above would not launch.

After much digging (here, discord, reddit), one of the comments indicated **vcdredist is needed**.  I didn't see this mentioned in the faq or installer alerting to it missing.

https://learn.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist