JamesCJ60/Universal-x86-Tuning-Utility #119: [BUG] 2.0.1 update loop

kilrah, 2023-08-14:
Installed from 2.0.1 releases

Home page shows 2.0.1

[image]

Settings however shows 2.0.0.24 and says an update is available

[image]

Download update reinstalls the same thing (actually only get options to Repair or Remove from the installer)

ensarisik, 2023-08-14:
An an other bug is this notification

[image]

kilrah, 2023-08-21:
Fixed in https://github.com/JamesCJ60/Universal-x86-Tuning-Utility/commit/1764b2268008d4ae796d2f9fac57919a9767a4e1

closed 2023-08-21 by kilrah