JamesCJ60/Universal-x86-Tuning-Utility #123: [BUG]ryzen 7840s settings not applying

craterboy1, 2023-08-25:
I just received the new ryzen 7840s for the Lenovo Yoga Slim 7. Any attempt to increase the TDP above factory doesn't apply at all. I understand on other laptops, vendor software would sometimes reapply settings. In this case, settings do not apply in the first place

Thanks for your time

craterboy1, 2023-08-25:
just to add to this, the TDP will boost to factory boost levels on high performance mode so I do not think there is TDP cap

Mach1212, 2023-11-17:
Same for Lenovo E14 Gen 3 5825u. Running the following command works though, 
```ryzenadj-win64\ryzenadj.exe --tctl-temp=95 --apu-skin-temp=95 --stapm-limit=35000  --fast-limit=40000 --stapm-time=64 --slow-limit=37000 --slow-time=128 --vrm-current=180000 --vrmmax-current=180000 --vrmsoc-current=180000 --vrmsocmax-current=180000

pause```

UXTU doesn't apply but Ryzenadj does.

closed 2023-11-25 by JamesCJ60