JamesCJ60/Universal-x86-Tuning-Utility #128: When trying to start the program it says "an  instance of Universal x86 Tuning Utility is already open!" and it actually already running on background, but I cannot open the program and setting it. pls fix.[BUG]

Lazulox, 2023-09-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS Version: [e.g. Windows 11 22H2]

**Additional context**
Add any other context about the problem here.

ruudvanstrijp, 2023-10-06:
I have the same issue. Somehow, the icon disappears from the system tray every now and then. I think it happens after hibernating the PC. So you see no icon and you think the app is not running, you want to start it but then apparently it already is running. The only way to open it, is by killing the process and starting the app again.

HongYue1, 2023-10-29:
I have the same problem, it disappears from the system tray.

alan8r, 2023-11-30:
same problem. win 11, after i suspend my laptop (single power button press, psuedo-sleep mode) then wake, the icon will sometimes not be in the system tray anymore but the program is still running according to task-manager and trying to re-run the program results in the alert saying the program is already running. 

program can be stopped via task manager and then re-ran but it would be nice if there was a way to recover the app in the tray without having to manually kill/restart

JamesCJ60, 2023-12-04:
This should now be fixed within the latest 2.2.2 build. Please disable and reenable the start on system boot option within the settings menu to apply the bug fixes!