JamesCJ60/Universal-x86-Tuning-Utility #130: [BUG]

Lindstroem4, 2023-09-23:
**Describe the bug**
Random system freeze and consequent crash with both 6800HS and 7840HS

**To Reproduce**
Steps to reproduce the behaviour:
Limit all TDP-settings to 30W (Autoreapply every 3s)
Use computer for 1-6 hours
System freezes. No such occurrences when not using the UXTU-settings.

**Expected behaviour**
No freezes/crashes :) 

**Screenshots**
N/A

**Desktop (please complete the following information):**
 Windows 11 23H2
Yoga Pro 7 7840HS
Yoga Slim 7 Pro X 6800HS

**Additional context**
Unsure if this is a known error, welcome any input and if it’s possible to alleviate the issue as I love the application and being able to set a fixed TDP-setting.

igornovitzky, 2023-09-23:
This reflects my experience on Lenovo z13 6850u

filius-martis, 2023-10-10:
Thinkpad P16s with 6850u and I see random system freezes, too, although I limit tctl_temp instead of TDP.

Note that this also happens with ryzenadj instead of UXTU, besides using one of those the system is rock stable.

Kr1s1m, 2023-11-11:
Same here with Lenovo Yoga Slim 7 Pro Ryzen 7 6800HS 16GB. But the freeze always happens while not under heavy load. So there is no freezes when I use it only during gaming with 28w power tuning custom preset and then not forget to override it by changing bios performance setting back to what it was and quitting the app.

Lahm919, 2023-11-12:
me toooooooooo.
what's more, my EVE v2 with i5-1135g7 can't use UXTU's adaptive mode at all. Cpu frequency and NB fre drops to 400MHz and locks.

JamesCJ60, 2023-11-12:
> Same here with Lenovo Yoga Slim 7 Pro Ryzen 7 6800HS 16GB. But the freeze always happens while not under heavy load. So there is no freezes when I use it only during gaming with 28w power tuning custom preset and then when done gaming not forgetting to override it by changing bios performance setting back to what it was and quitting the app.

IGPU driver instabilities are causing the issues you have experienced here relating to the 6000 series and newer APUs. This has been reported to AMD, who were already aware and are looking into the matter.

filius-martis, 2023-11-14:
Do you have a link to AMD`s statement on the issue?

For me it looks like the freezes might occur when reaplying settings via UXTU or ryzen-adjust collides with my vendors baked-in routines (UEFI?!)  that I unfortunately cannot disable. 
Therefore it helps to have a larger reapply-timeout, 20s or more and the freezes are almost gone here.

Kr1s1m, 2023-11-14:
> Same here with Lenovo Yoga Slim 7 Pro Ryzen 7 6800HS 16GB. But the freeze always happens while not under heavy load. So there is no freezes when I use it only during gaming with 28w power tuning [...]

Also, I want to add that during gaming using UXTU and limiting to 28w sometimes extremely rarely causes these sudden fps drops to like 2 fps which continue for a few seconds and then disappear for a very long while. Do you think it could be related to boost duration somehow? Or maybe something else? I do not get the random rare drops in fps when I do not use the application (but my laptop is ultra thin so it heats up too much then, if I don't use UXTU to effectively turn the HS into an U, so I just deal with the occasional drops).

Kr1s1m, 2023-11-14:
> IGPU driver instabilities are causing the issues you have experienced here relating to the 6000 series and newer APUs. This has been reported to AMD, who were already aware and are looking into the [...]

I see. I will be updating drivers and UXTU in the future, hoping that it gets resolved. Will report if it starts working normally on 6800 HS.

JamesCJ60, 2023-12-02:
If anyone wishes for a solution for now, you have 1 of 3 options:
- Increase the Radeon drivers sensor metrics polling rate
- Use the minimal install for the Radeon GUI
- Use the driver only install option

Dave9009, 2024-02-08:
Version V2.2.4 fixed random freeze?  The same thing happened to me with previous versions. CPU: 7735hs

SWi7CHbladE, 2024-02-16:
I am facing the same issue with my ryzen 7 7735u on lenovo yoga 7 14arp8. Driver-only GPU driver installation just delays the issue.

MetalRex101, 2024-03-15:
Ryzen 7 7745HX - the same problem. I've limited CPU temps to 70 degrees and started to get driver BSOD, crashes right after startup (with config auto reaply every 5 seconds and on startup), complete power off.

closed 2025-06-23 by JamesCJ60