JamesCJ60/Universal-x86-Tuning-Utility #131: [FEATURE REQUEST] Dragon Range

SoonpilHOng, 2023-09-30:
Is there no possibility of supporting for the Dragon Range CPUs at all?

cidiousx, 2023-10-19:
It's the AMD CPU that needs this tool the most. But the dev doesn't seem to be interested at all to add it. Keeps saying due to back-end rewrite for months now. I have given up hope. It is what it is.

ASURA715, 2023-10-20:
It's the AMD CPU that needs this tool the most

JamesCJ60, 2023-12-02:
At this current time, no one is aware of the addresses needed to communicate with Dragon Range’s SMU.

closed 2023-12-02 by JamesCJ60

fscottcopeland, 2023-12-29:
@JamesCJ60 see https://github.com/irusanov/SMUDebugTool - I've got per core CO working on my 7945HX using this tool.

fscottcopeland, 2024-03-09:
> @JamesCJ60 see https://github.com/irusanov/SMUDebugTool - I've got per core CO working on my 7945HX using this tool.

Reminder @JamesCJ60