JamesCJ60/Universal-x86-Tuning-Utility #134: Screen Resoulution changes to 640x480 when applying custom preset with refresh rate, will not go back to original resoulution unless utxu is closed

HackifyPhython, 2023-10-12:
**Describe the bug**
The screen resoulution changes to 640x480 when applying the custom preset with refresh rate at 60Hz. It will not go back to the original resoulution unless I end the task of utxu or close it. I think it has something to do with the refresh rate. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'custom preset'
2. Click on 'apply preset'
4. See error

**Expected behaviour**
The screen resoulution automatically changes to 640x480 and i can't see shit because too much is crammed into the small screen resoulution.

**Desktop (please complete the following information):**
 - OS Version: Windows 10 Home 22H2

JamesCJ60, 2023-10-14:
This should be fixed as of the latest build.

closed 2023-10-14 by JamesCJ60

advait404, 2024-03-20:
@JamesCJ60 This issue has resurfaced. Tested on V2.3.0 - Beta 1 and V2.2.5 Stable.