JamesCJ60/Universal-x86-Tuning-Utility #136: [BUG] Updating creates new shortcut even if it already exists

knasiotis, 2023-10-20:
I do not know if this has been fixed, however I just updated to version 2.1.0.0 and it created a second identical shortcut. I believe this is probably trivial to fix, just a small qol change to avoid removing the redundant shortcut manually.
Thanks in advance!

LiberteCzech, 2023-10-20:
I upgrade from 2.1.0 to 2.1.1 and no extra shortcut, so it is propably working.

knasiotis, 2023-10-20:
are you using windows 11?

LiberteCzech, 2023-10-20:
Yes I use the latest 11 stable.
[image]

knasiotis, 2023-10-20:
same here, weird, i've been having the same issue on multiple versions not only 2.1.1

JamesCJ60, 2023-10-20:
Hi, the installer package is maintained by Microsoft and any bugs have to be addressed by them.

closed 2023-10-20 by JamesCJ60