JamesCJ60/Universal-x86-Tuning-Utility #145: [FEATURE REQUEST]

FNsi, 2023-11-06:
**Is your feature request related to a problem? Please describe.**

No

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

DDR overclocking

**Describe alternatives you've considered**

 SPD backup, edit and flash...

**Additional context**

I think... it's hard but really useful in any case...

squee666, 2023-11-11:
wouldnt be feasible its hard defined by bios
if you need to do that closest you can do is use smokeless UMAF and do it yourself https://github.com/DavidS95/Smokeless_UMAF

FNsi, 2023-11-12:
> wouldnt be feasible its hard defined by bios/uefi
> 
> if you need to do that closest you can do is use smokeless UMAF and do it yourself https://github.com/DavidS95/Smokeless_UMAF

Nah, I don't think that works universally. 

Meanwhile I can do it by setup_var manually, but that's complicated to use.

closed 2023-11-25 by JamesCJ60