JamesCJ60/Universal-x86-Tuning-Utility #146: [BUG] - Adaptive iGPU clock stays stuck at 400mhz/ desktop version and 835mhz/ handheld version - 6800u

dutch45612, 2023-11-09:
**Describe the bug**
Hello, i looked and didn't find the answer here... Well i tried both version, and the adaptive function on both stuck the iGpu frequency to those mentioned above, the usage raise to 100% rapidly and the clock stays there, so the fps drops. I don't know if it is a known issue but this is it. My system is a Winmax 2 2022 (6800u) and i am on Windows 11 22H2. - Thanks for the program it is really versatile, i would really like to use it if the adaptive iGpu could works as intended :).

**To Reproduce**
Start adaptive mode with any minimum iGpu frequency

**Expected behaviour**
Automatically manage iGpu frequency according to load

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 Windows 11 22H2

anoosi, 2023-11-26:
same thing 7735hs Ser6 Pro mini pc

JamesCJ60, 2023-11-27:
Thanks for the report. I am aware of this issue. The issue may be from ADLX from some brief debugging, so I will aim to fix it once I get some free time over the next few weeks.

JamesCJ60, 2023-12-30:
This should now work as of the latest release.

closed 2023-12-30 by JamesCJ60

michaelcrusso, 2024-07-22:
I'm still experiencing this issue on my AMD Ryzen 7 8840U w/ Radeon 780M Graphics, on Windows 11 23H2.

Adaptive Mode with the Turbo Boost Overdrive iGPU settings essentially breaks the machine -- it sets the GPU clock to the lowest permitted clock (so if I increase minimum clock to 600, it'll be stuck there) and nothing short of a machine restart will fix it, not even disabling Adaptive Mode and trying to force apply a normal custom preset (if I set a flat frequency through the Custom window, and then disable it, it will eventually drop back down to Adaptive Mode's lowest frequency). I also notice extremely high power usage (GPU ASIC Power in HWInfo64) despite the fact that it's stuck at 400 MHz, and my computer's fans go haywire.

BertrandBordage, 2024-09-14:
Same issue here with a Zenbook S 16 with a Ryzen AI 9 HX 370 & Radeon 890M.

Once you activate once the adaptive mode with turbo boost overdrive, **nothing can turn the iGPU clock back to its factory state**. In my case, it stays at exactly 160 MHz when it was usually idling between 600 and 1000 MHz.

I tried reinstalling AMD drivers, reinstalling MyAsus, tweaking/toggling settings with GHelper, MyAsus, UXTU, rebooting multiple times, this never changes. The only "solution" for me is to have UXTU running at all time with adaptive mode except… that due to another bug, the CPU clock is underclocked to 600 MHz instead of its default 1600-2000 MHz idle state. Which makes the entire OS extremely laggy :(
[image]

m00vie, 2024-10-28:
Same on 6800H radeon 680m