JamesCJ60/Universal-x86-Tuning-Utility #151: Please give me a link to the NET framework I need.

Wozza2014, 2023-11-29:
Hi.

The Universal-x86-Tuning-Utility installer opens a page with many different NET installers.  There is SDK, ASP.NET Core Runtime 8.0.0, .NET Desktop Runtime 8.0.0, .NET Runtime 8.0.0.  

Which one do I need to download??

I've installed ASP.NET Core Runtime 8.0.0 and also tried .NET Runtime 8.0.0 and its still asking for NET framework when I try to install Universal-x86-Tuning-Utility.

Please can you give me a link to the exact one.

Thanks.

closed 2023-12-01 by JamesCJ60

Wozza2014, 2023-12-02:
Hi.

Thanks for that.  One more thing.  Does the runtime need to be the x86 version?  I take it the apu tuning app is 32-bit due to the x86 being in the name?

JamesCJ60, 2023-12-02:
You need the 64bit version.