JamesCJ60/Universal-x86-Tuning-Utility #155: [BUG] 'Provide value on System.Windows.Bam12006.TypeConverterMarkupExtension' threw an exception.' Line number '86' and line position '13'.

NextDev65, 2023-12-05:
**Describe the bug**
Program will not launch, error popup appears instead.

**To Reproduce**
Steps to reproduce the behaviour:
1. Install
2. Run
3. Admin UAC > Yes
4. Error Popup

**Expected behaviour**
App Window (UI) should launch.

**Screenshots**
[image]

**Desktop:**
 - Windows 10 Home 22H2

**Additional context**
The exe is running in task manager but there is no window on systray icon.
Same error on v2.2.2 or v2.2.1

ObservantRook, 2024-09-20:
i got the same problem, last time i fixed it with a factory reset but i don't really want to try that anymore

NextDev65, 2025-08-21:
Portable version is working fine on the latest. Cheers!