JamesCJ60/Universal-x86-Tuning-Utility #161: [BUG] Custom Preset not applied at start

hardysim, 2023-12-29:
**Describe the bug**
I have made a custom preset (with core offset) and saved it. I have enabled start on boot and reapply on start. The app is started on boot but my custom preset is not applied.

**To Reproduce**
Steps to reproduce the behaviour:
1. create custom preset and save it
2. enable reapply on start
3. restart app / PC

**Expected behaviour**
The custom preset should be applied on start

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
- Windows 11 23H2
- Universal x86 Tuning Utility 2.2.2.0

**Additional context**
Add any other context about the problem here.

JamesCJ60, 2023-12-29:
Your APU does not support Curve Optimiser.

closed 2023-12-29 by JamesCJ60

hardysim, 2023-12-29:
Oh really? It does run noticeable faster when I set a negative offset tough. So it does work / make a difference?

But I have to set it manually and wanted to ask if the app can set it automatically.

JamesCJ60, 2023-12-29:
Curve Optimiser only works for Ryzen 9 APUs. It was just a placebo.

hardysim, 2023-12-29:
> Curve Optimiser only works for Ryzen 9 APUs. It was just a placebo.

Does this apply only for mobile processors? It works for a ryzen 7 desktop, right?

But it's really weird because I really can reproduce a better performance and higher clocks when setting the offset.

Never mind. It was worth a try. Thank you very much.