JamesCJ60/Universal-x86-Tuning-Utility #174: [BUG] Custom Auto Reapply interval is not preserved between app restarts

xak2000, 2024-03-04:
**Describe the bug**
The setting `Auto Reapply (s)` allows to set custom interval (default is `3`), but whatever value is set here, after app is restarted, the value returns to `3`.

**To Reproduce**
Steps to reproduce the behaviour:
1. Start the app.
2. Go to `Settings`.
3. Enable `Auto Reapply (s)` and set custom interval to `60`.
4. Close the app.
5. Start the app.
6. Go to `Settings`.
7. Check the interval. It is `3`.

**Expected behaviour**
The interval is preseved between app restarts. So, after restart it should be `60`.

**Desktop (please complete the following information):**
 - OS Version: Windows 11 22H2.
 - UXTU version: 2.2.5.0.

**Additional context**
Legion 5 Pro with 6800H/3070.

realNZZN, 2024-06-15:
This has been the case since the rewrite, always defaults to 3 regardless of what it was set to before.

Aleksis77, 2025-02-23:
The same problem - the specified value in the Autoreapply function is not saved after restarting UXTU. This has been happening for a long time, since AATU, Please fix this bug. I really need it.
Tested on two different laptops with 4900HS and 7945HX

Also, this Autoreapply feature has some weird interaction with some apps. For example I am using NDI Screen Capture and when Autoreapply is running, every 3 sec. blinks in the captured Screen Capture window.
Back in the AATU days, this feature used to work even weirder - various elements in the application windows would blink all the time. So I had to turn it off. Nevertheless, the Autoreapply feature is needed and a big request to finalize it. Thank you!

vardin23, 2025-06-22:
Issue still not fixed
Happens on both versions

vardin23, 2025-06-23:
This issue still persists for me unfortunately.
Auto reapply resets to 3 on every start after exit.
OS : Windows 11 24H2 Latest Build

I've checked multiple time intervals so I don't think its a problem with the intervals.
I believe it's tied to the preset in some way so every time the preset is applied it resets to 3.
Also , I can't find any specific cfg file or anything where one can just edit to make these changes persistent.

ozhasta, 2025-06-26:
Hello, seems still persist on v2.5.3.0 both installed and protable versions.

Aleksis77, 2025-06-28:
Yes, everything is true - this is preserved from version to version. Therefore, I try not to enter the settings once again.)) So that the counter does not drop 3 seconds by default. It is probably difficult to fix it - if the developer is knowing about it, leaves everything as it is.