JamesCJ60/Universal-x86-Tuning-Utility #178: [FEATURE REQUEST] Support for Dragon Range CPU - 7945HX

antarct, 2024-03-19:
I own MSI Alpha laptop with 7945HX CPU. Is there any chance that your application with support this machine?
Currently it's very loud and hot - no matter what I change min clocks are above base clocks and the power draw is really high.

MetalRex101, 2024-03-20:
@antarct I would advice you to start with disabling CPU turbo boost with Windows power plan settings. You will need first to make this setting visible through registry edit or cmd command. This will already reduce your CPU temperature a lot. If you don't do any cpu-heavy work or don't expect to play on ultra preset with cap fps, it will do the trick for you. I have Ryzen 7 7745HX.

I also tried to keep CPU turbo boost and limit CPU frequence up to 4GHz, but CPU keeps to get 4.6GHz instead. The best would be of course to just limit power up to 30-35w, but once I try to change power settings with UXTU, I get system crashes or stuck.

antarct, 2024-03-20:
Thanks for answer.
My problem is that I really need full potential of this CPU about half of the time I work. I'm looking for a way to make it colder/quieter when it's not fully loaded.
If I only knew that mobile AMD processor consumes a lot more power when idle than 13700k...
I've tried limiting this one to eco mode/65W but then I don't get full performance when I need it.
After some measurements I think that high idle voltage is the problem, it's more than 1.3V when the CPU is doing nothing, probably because of thermal headroom..

MetalRex101, 2024-03-20:
@antarct dragon range line unfortunately has very bad software support so far.. As far as I know Phoenix range (HS) CPUs have much better Idle consumption, because they are made as a single chip, not a chiplet like HX series. In fact, HX is a desktop CPU with a different label. On the other side, intel CPUs have much higher TDP for almost the same performance. You can compare those parameters on [notebookcheck.com ](https://www.notebookcheck.net/), they usually have quite good reviews.

Edensystem, 2024-05-28:
On my msi c7vg with a 7945HX, the UXTU settings that work are the curve optimiser and the « amd boost profile ».
All the CPU power settings such as TDP PPT STAPM PBO etc etc .. don’t have any effect.

Even in the advanced UEFI menu (ALT ALT SHFT F2 key) in the SMU tab, changing values don’t affect cpu behaviour, but sometimes on the first save and reset you get full performance, kind of a SMU bug ….
I also tried the [UEFI var utility](https://github.com/datasone/setup_var.efi) to access overclocking variable, with sometimes the same SMU bug where you get higher values in the STAPM settings (seen in RWeverything) , but everything get back to the factory behaviour on next reboot, while uefi var are still modified.

There is a tool from irusanov called Zenstates where you can adjust the « Fast PPT Limit » with the « PPT » parameter, and another called SMU dedug tool, where a lots of parameters are tweakable, but there isi no documentation to use it properly.

There is another tool called [UMAF](https://github.com/DavidS95/Smokeless_UMAF) (Universal Amd Form Browser) where you can access to the overclocking tab with a UEFI menu, change the curve Settings precisely and save it , but you will lose the UEFI firmware update option in the stock UEFI menu, it happened to me … only a new firmware detected by windows update saved me by update the firmware automatically. 😅

Anyway, the safest way I found was with UXTU,  you can adjust the curve optimiser with the first 8 core, and also the other 9to 16 cores with the global curve optimiser . You cannot achieve a fine tuning, but temperature drop significantly while overall performance get a bump.

antarct, 2024-05-29:
> On my msi c7vg with a 7945HX, the UXTU settings that work are the curve optimiser and the « amd boost profile ». All the CPU power settings such as TDP PPT STAPM PBO etc etc .. don’t have any effect.

What are your settings?

Edensystem, 2024-05-29:
My setting for global curve optimiser is -19  ( CCD1 and CCD2) And for the « per core curve optimiser (CCD1) » -4 for the 2 best core , -3for the next 2 etc etc
After a few freeze, this is the best safe settings for my CPU.
At first , I have been able to go at -29 on the global setting, (without using per core tab), but it was not really stable….

You can try -20 on global curve optimser and start from there to see if your CPU can handle more without any glitches..

Edensystem, 2024-06-04:
For MSI C7V with 7945HX, there is a recent  [bios]([url](https://www.msi.com/Laptop/Alpha-17-C7VX/support?sub_product=Alpha-17-C7VG)) update that fix user settings in the SMU tab.
Now when you set user scenario to « performance mode »  , SMU settings are taken into account.

antarct, 2024-06-04:
> For MSI C7V with 7945HX, there is a recent bios update that fix user settings in the SMU tab. Now when you set user scenario to « performance mode » , SMU settings are taken into account.

It really does work. Thanks for the info.

antarct, 2024-06-06:
@Edensystem Have you tried disabling iGPU after switching to Discrete Mode? I thought it will take some power from the CPU but when I disable iGPU laptop behaves strange - it doesn't show charging icon on battery.

Edensystem, 2024-06-26:
> @Edensystem Have you tried disabling iGPU after switching to Discrete Mode? I thought it will take some power from the CPU but when I disable iGPU laptop behaves strange - it doesn't show charging icon on battery.

I have tried on a previous version of firmware 3 month ago, and get a dark screen on reboot, so had to remove the battery to reset the UEFI settings... never tried again,. but there is a slight CPU performance increase just by switching on discrete mode, it seems  the package drain less power.

MidnightCrazy, 2024-08-21:
> My setting for global curve optimiser is -19 ( CCD1 and CCD2) And for the « per core curve optimiser (CCD1) » -4 for the 2 best core , -3for the next 2 etc etc After a few freeze, this is the best [...]

@Edensystem Excuse me, what is the version of the UXTU, I installed the 1.0.3Stable, but I didn't find any settings for per core curve optimiser...

el-banan, 2024-09-30:
@MidnightCrazy @Edensystem what version of UXTU were you using to make the adjustments?

LHX369963, 2024-10-24:
try msi center. user scenario-extreme performance-advanced settings-fan speed-advanced

laptop-abuser, 2025-05-08:
@Edensystem disabling igpu from bios leads to this weird bug, but if you disable it from windows device manager then cpu pulls ~56w instead of 50w while gaming. I got nice increase in performance in cpu dependent games, like cs2.

closed 2025-06-23 by JamesCJ60