JamesCJ60/Universal-x86-Tuning-Utility #179: [FEATURE REQUEST] Linux support?

OzzyHelix, 2024-03-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I use Linux and I can't use this because its a Windows application 
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
it would be cool to have a port to linux 
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
ryzen-controller is deprecated so something like this on linux would be cool
**Additional context**
Add any other context or screenshots about the feature request here.

DoctorKM, 2024-03-20:
I'm ready for testing, if that would be needed. (Arch, AMD Ryzen 9 7940HS)

MestreLion, 2024-03-31:
I also volunteer for testing!

2 Desktops running Ubuntu 18.04, 22.04 and 24.04, AMD Ryzen 5 2600 and Ryzen 7 5700G

gorouflex, 2024-04-17:
You can try [UXTU4Unix](https://github.com/AppleOSX/UXTU4Unix)

arabcian, 2024-05-04:
Can anyone test if uxtu4unix lets you undervolt dragon range cpus?

gorouflex, 2024-05-04:
> Can anyone test if uxtu4unix lets you undervolt dragon range cpus?

No, Dragon ( or Dragon Range ) are not supported by RyzenAdj

arabcian, 2024-05-13:
At least can we have which msr addresses responsible for curve optimizer for dragon range cpus?

Splarkszter, 2024-06-28:
There is currently no GUI maintained for Linux :(

Well. At least what i use (RyzenAdj) isn't hard to use on the CLI but took a while to understand what each setting does, and i don't want to break my system :)

teotikalki, 2024-07-16:
May I humbly suggest that this project change it's name to "Windows x86 Tuning Utility".
As a Windows-only  program it is FAR from 'Universal'.

kikislater, 2024-07-17:
And .net is a mess in Linux ! I don't understand why people are still using it ... 
So the comment is on the website makes me laughing 

> "The development of this utility is spearheaded by the same talented team that brought us popular applications like [Ryzen Controller](https://gitlab.com/ryzen-controller-team/ryzen-controller), [Renoir Mobile Tuning](https://github.com/sbski/Renoir-Mobile-Tuning), and [Power Control Panel](https://github.com/project-sbc/Power-Control-Panel-v2)."

HakaishinShwet, 2024-08-17:
:-((

mingshi2333, 2024-09-06:
I also volunteer for testing 
  from 7745hx fedora

Splarkszter, 2024-09-09:
RyzenAdj works perfectly. I've set batch scripts to set "power profiles" depending on what I want to do, either performance or battery save. Really good!

OzzyHelix, 2024-09-09:
Linux doesn't have CPU Overclocking tools like Windows does Ryzen Master and Universal X86 Tuning Utility don't exist on Linux my hope was that by making this issue there might be some interest in porting Universal X86 Tuning Utility to Linux so we Linux users can have a design UI for Overclocking

gorouflex, 2024-09-09:
You can try [UXTU4Unix](https://github.com/AppleOSX/UXTU4Unix), although it only supports CLI for noơ 🐧🛐

gorouflex, 2024-09-09:
> one thing has changed between when I made this thread and now. I upgraded from a Ryzen 7 5700G to a Ryzen 9 7900X3D so my interest in this post is more for getting a port for other Linux users who want a GUI for CPU Overclocking I thought porting this app might be viable

I know but I'm rarely have free time to dev my project

OzzyHelix, 2024-09-09:
I understand. I'm just making a case for a port I'm in no hurry for it to happen and I am sorry if I came off as being a bit pushy

arabcian, 2024-09-10:
> I understand. I'm just making a case for a port I'm in no hurry for it to happen and I am sorry if I came off as being a bit pushy

Can we at least have 7745 7845 7945 HX undervolt support on linux? I tried JamesCJ's pull request but it causes a hang on ryzenadj.

arabcian, 2024-09-10:
> I know but I'm rarely have free time to dev my project

Can we at least have 7745 7845 7945 HX undervolt support on linux? I tried JamesCJ's pull request but it causes a hang on ryzenadj.

Vistaus, 2024-10-09:
As a Linux user I would love to see this app on Linux as well.

JamesCJ60, 2025-01-03:
Hi all, I cannot get into much detail right now, but UXTU’s successor, which I have started development work on, aims to support Linux (provided nothing goes wrong) and will merge a few of my projects into one (e.g., UXTU, UXTU Handheld, Framework Hub, etc.).

gorouflex, 2025-01-03:
> Hi all, I cannot get into much detail right now, but UXTU’s successor, which I have started development work on, aims to support Linux (provided nothing goes wrong) and will merge a few of my projects into one (e.g., UXTU, UXTU Handheld, Framework Hub, etc.).

That's kinda so I'll only focused on development on macOS/Hackintosh

gorouflex, 2025-01-03:
@JamesCJ60 I'll deprecate the Linux version for my UXTU4Unix project then...

GuavaEater, 2025-02-21:
Let me know if you need a tester for the Linux version, I'll happily help out.

laptop-abuser, 2025-06-02:
@JamesCJ60 Is there any progress on a linux version? I'm using your software for a year now on windows and it's great. Would be very nice to get linux support.

JamesCJ60, 2025-07-29:
Hi all! Just wanted to keep you all in the loop. @loyvsc is currently working on a Linux port of this version of UXTU. It’s a work in progress, so we appreciate your patience and support. Again, big thanks to @loyvsc for taking this on!

a137kate, 2026-08-28:
@JamesCJ60 

Hey!
Thank you for your work!
Please let me know when you need volunteers for testing