JamesCJ60/Universal-x86-Tuning-Utility #180: [FEATURE REQUEST] Portable Application?

gebbet, 2024-03-20:
If possible, could you please make UXTU a portable application which would not require an installation?

dev-bd, 2024-03-25:
I can potentially try and repack it for you, depending on how the application functions it shouldn't be too difficult.

gebbet, 2024-03-25:
> I can potentially try and repack it for you, depending on how the application functions it shouldn't be too difficult.

I have resolved my installer issue in safe mode but I was asking for the feature for general ease of use. The most of the apps/tools on my system are portable for example; it is easier to move them between different systems while keeping the settings if needed, and to be able to use them with just a click is great in my opinion.

closed 2025-06-23 by JamesCJ60