JamesCJ60/Universal-x86-Tuning-Utility #185: [BUG] Do not start

Honusnap, 2024-03-31:
**Describe the bug**
Application window appear then diseapear 

**To Reproduce**
Steps to reproduce the behaviour:

Windows 11
Install last stable from the website, asked for windowsdesktop-runtime-7-win-x64, installed => Run application, show a window for half a sec then diseapear => No error message, application crashed
Came to github, downloaded latest 2.2.9, asked for windowsdesktop-runtime-8.0.3-win-x64, installed => same problem

**Expected behaviour**
Well ... the windows application stay opened... that i can use it .. :D

**Screenshots**
None possible .. appear too fast

**Desktop (please complete the following information):**
 - OS Version: Windows 11 22H2 - Family

**Additional context**
Add any other context about the problem here.

xxSENPAIxx, 2024-04-02:
Same Issue for me. Have you found a solution?

Honusnap, 2024-04-05:
> Same Issue for me. Have you found a solution?

Hi, nope... and i tried the "run as admin", not better.

offthewallds, 2024-05-13:
> Same issue. In my case, installing Adrenaline software from AMD solved the problem.

This solved the issue for me. On a new Envy x360 with AMD 8840HS. Before installing UXTU I uninstalled all HP software including their "tuning" application. I tried going back to 2.2.1 as mentioned in the other "no start" thread. No go. Installing Adrenaline software worked immediately.

Wenbobobo, 2024-08-05:
me too. But it still cannot start after install AMD Adrenaline software