JamesCJ60/Universal-x86-Tuning-Utility #186: [BUG] App opens only for a brief moment.

xxSENPAIxx, 2024-04-02:
**Describe the bug**
The application opens the window only for a brief moment and closes.

**To Reproduce**
Steps to reproduce the behavior:
1. Double clicking on app
2. wait for it to load
3. see a flash of the window
4. closes very quickly (under second)

**Expected behaviour**
Window stays open, application stays running.

**Screenshots**
Happens too fast to take a screen shot.

**Desktop (please complete the following information):**
 - OS Version: [e.g. Windows 11 23H2]
 - Asus Zephyrus G14 2022 , Ryzen 6900hs and Radeon 6700s

**Additional context**
Add any other context about the problem here.

1MrEnot, 2024-04-05:
Do you see UXTU window or something else? You could try to use snipping tool (win+shift+s) and record a video
It may be somehow related to administrator pemissions. UXTU does check admin permissions and restart if necessary, but I would suggest you run it as administrator manually. On my machine shortcut does not provide "run as administrator" option, but .exe does

Honusnap, 2024-04-06:
> Do you see UXTU window or something else? You could try to use snipping tool (win+shift+s) and record a video It may be somehow related to administrator pemissions. UXTU does check admin permissions [...]

I got the same problem, yes the window appear for like .. half a sec then go away.
Running as admin change nothing.

1MrEnot, 2024-04-07:
I added logging to file, so it will be easier to troubleshoot. Now you can wait for next beta release and download it or build UXTU locally. Then run it and send log file here, it will help a lot

To built app locally you have to:
- download and install .NET 8 SDK [here](https://dotnet.microsoft.com/en-us/download/dotnet/thank-you/sdk-8.0.203-windows-x64-installer)
- clone this repository
- go to cloned repo and run these commands
```
dotnet build
dotnet run --project 'Universal x86 Tuning Utility'
```
Log file will be in `./logs/` and called something like `uxtu_log20240408.txt`

Honusnap, 2024-04-08:
> I added logging to file, so it will be easier to troubleshoot. Now you can wait for next beta release and download it or build UXTU locally. Then run it and send log file here, it will help a lot [...]

I will wait next Beta and supply the logfile aftwerward.

guilamu, 2024-04-24:
Hello,

Exact same issue here, and I tried the last 4 versions (3 betas and 1 stable).