JamesCJ60/Universal-x86-Tuning-Utility #189: How to remove all applied configuration

diotriandika, 2024-04-09:
I've uninstalled UXTU from my device, but idk how to completely remove applied preset after uninstalling it. The major problem i have were at GPU TDP, im using ghelper and usually it runs at 80w at balance mode. But now it start kicking to 100w even 120w. I've tried to restart my computer but still not working.
[image]

JamesCJ60, 2024-04-09:
Hi, UXTU will not and cannot keep presets applied across system boots nor control GPU power limits.

closed 2024-04-09 by JamesCJ60

diotriandika, 2024-04-10:
Im sorry, but it does 
I have accidentally configured in the Nvidia section on the newest UXTU. After that my gpu TDP Uncapped even in balance mode. In normal situations it should be limited on 80watt, then i reset my laptop and the gpu still uncapped
Please help me, it makes my computer runs hotter rn

JamesCJ60, 2024-04-10:
Again, there is zero code within UXTU to control the TDP of NVIDIA GPUs.