JamesCJ60/Universal-x86-Tuning-Utility #202: [BUG]UTU opens and closes in a flash, doesnt even appear in taskbar

ModeusRAT, 2024-05-23:
**Describe the bug**
UTU doesnt run in the background, even when the program is repaired or uninstalled and installed again
**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'Universal Tuning Utility'
2. Start the program as admnistrator
3. Wait and see program open and close quickly
4. UTU doesnt run 

**Expected behaviour**
Don't really know, used to work before but now doesnt

**Desktop (please complete the following information):**
 - OS Version: 22631.3593
 - CPU: Ryzen 5 5500U
 - Ram: 8gb of ram

Varusal, 2024-08-28:
Have the same issue.

OS Version: Windows 11 Pro - 22635.4802
CPU: Ryzen 5 7530U
Ram: 32GB

m-aar-ten, 2024-08-31:
> Have the same issue.
> 
> OS Version: Windows 11 Pro - 22635.4802 CPU: Ryzen 5 7530U Ram: 32GB
> 
> EDIT: Solved by uninstalling drivers and reinstalling them.
Which drivers did you uninstall? I am running into the same problem.

Varusal, 2024-09-05:
> > Have the same issue.
> > OS Version: Windows 11 Pro - 22635.4802 CPU: Ryzen 5 7530U Ram: 32GB
> > EDIT: Solved by uninstalling drivers and reinstalling them.
> 
> Which drivers did you uninstall? I am running into the same problem.

I reinstalled the chip set and graphics drivers.