JamesCJ60/Universal-x86-Tuning-Utility #210: [BUG] Latest release is asking for .Net 7.0 (I have 8.06 installed)

mooms06, 2024-06-20:
**Describe the bug**
I have UXTU 2.3.0b4 (2.2.10) installed, when I try to upgrade to beta 4 I have this popup:
[image]

 - OS Version: [e.g. Windows 11 23H2]

mooms06, 2024-07-22:
Still not fixed in beta 6.
I had to install .NET 7.0 Desktop Runtime v7.0.20 to be able to install UXTU.

Once UXTU installed, .Net 7. can be uninstalled, so the issue probably lies in the installer of UXTU that haven't been updated to check for .Net 8 instead of .Net 7.

mooms06, 2024-08-18:
Still not fixed in 2.3.1 stable....

[image]

Dyminas, 2024-09-22:
The installer still requires .NET 7.0 in 2.3.2 stable...
[image]

kg, 2024-09-25:
My sister installed the 7.0 installer and then got this error when launching tuning utility
[image]
What gives? Does it require both 7 and 8?

mooms06, 2024-10-06:
I've made a clean install of Windows 11 24H2, and .Net 7.0  wasn't asked by the installer, so maybe the solution is to uninstall UXTU completely before installing a newer version.

JamesCJ60, 2026-02-20:
Hi, this issue has been fixed as of the latest beta builds.