JamesCJ60/Universal-x86-Tuning-Utility #220: [BUG] WPFUI bug preventing launch

timber-schroeder, 2024-07-24:
**Describe the bug**
`System.IO.FileNotFoundException: The system cannot find the file specified. (0x80070002)`
is thrown at:
`Wpf.Ui.Interop.Dwmapi.DwmGetColorizationParameters(DWMCOLORIZATIONPARAMS& dwParameters)`

Full trace:
```
Application: Universal x86 Tuning Utility.exe
CoreCLR Version: 8.0.724.31311
.NET Version: 8.0.7
Description: The process was terminated due to an unhandled exception.
Exception Info: System.IO.FileNotFoundException: The system cannot find the file specified. (0x80070002)
   at Wpf.Ui.Interop.Dwmapi.DwmGetColorizationParameters(DWMCOLORIZATIONPARAMS& dwParameters)
   at Wpf.Ui.Interop.UnsafeNativeMethods.GetDwmColor()
   at Wpf.Ui.Appearance.Accent.GetColorizationColor()
   at Wpf.Ui.Appearance.Theme.Apply(ThemeType themeType, BackgroundType backgroundEffect, Boolean updateAccent, Boolean forceBackground)
   at Wpf.Ui.Appearance.Watcher.UpdateThemes(SystemThemeType systemTheme)
   at Wpf.Ui.Appearance.Watcher.<>c__DisplayClass12_0.<Watch>b__0(Object sender, RoutedEventArgs args)
   at System.Windows.EventRoute.InvokeHandlersImpl(Object source, RoutedEventArgs args, Boolean reRaised)
   at System.Windows.UIElement.RaiseEventImpl(DependencyObject sender, RoutedEventArgs args)
   at System.Windows.BroadcastEventHelper.BroadcastEvent(DependencyObject root, RoutedEvent routedEvent)
   at System.Windows.BroadcastEventHelper.BroadcastLoadedEvent(Object root)
   at System.Windows.Media.MediaContext.FireLoadedPendingCallbacks()
   at System.Windows.Media.MediaContext.FireInvokeOnRenderCallbacks()
   at System.Windows.Media.MediaContext.RenderMessageHandlerCore(Object resizedCompositionTarget)
   at System.Windows.Media.MediaContext.RenderMessageHandler(Object resizedCompositionTarget)
   at System.Windows.Media.MediaContext.Resize(ICompositionTarget resizedCompositionTarget)
   at System.Windows.Interop.HwndTarget.OnResize()
   at System.Windows.Interop.HwndTarget.HandleMessage(WindowMessage msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Interop.HwndSource.HwndTargetFilterMessage(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   at MS.Win32.HwndSubclass.DefWndProcWrapper(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
```
```
Faulting application name: Universal x86 Tuning Utility.exe, version: 2.2.13.0, time stamp: 0x661f0000
Faulting module name: KERNELBASE.dll, version: 10.0.22621.3880, time stamp: 0xdb9989e8
Exception code: 0xe0434352
Fault offset: 0x000000000005fabc
Faulting process id: 0x0xD82C
Faulting application start time: 0x0x1DADE1AA9FA4E1B
Faulting application path: C:\Program Files\JamesCJ60\Universal x86 Tuning Utility\Universal x86 Tuning Utility.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: e1d219f0-1d60-44a7-9421-0b8691816253
Faulting package full name: 
Faulting package-relative application ID: 
```

The application never launches it's GUI, it just crashes silently. I got this error from the event viewer.

**To Reproduce**
1. Launch the application.

**Expected behaviour**
For the application to launch.

**Desktop (please complete the following information):**
Win 11 10.0.22631.3880

**Additional context**
This is caused by a bug in WPFUI. Many other projects have run into this issue, a big example being Power Toys. A newer version of WPFUI (>=`3.0.0-preview.12`) fixes it.
See:
* https://github.com/microsoft/PowerToys/pull/30411
* https://github.com/lepoco/wpfui/pull/859

This project appears to have 2.1.0 pinned, which is quite old: https://github.com/JamesCJ60/Universal-x86-Tuning-Utility/blob/2b8fc50404828f0d3b3f7558ded94d3b54427617/Universal%20x86%20Tuning%20Utility/Universal%20x86%20Tuning%20Utility.csproj#L51

Bukollah, 2025-05-29:
I have the same issue. Since I am not a computer science major I didn't understand any of these. I have download the "Universal x86 Tuning Utility V2.5.1". Now what do I need to make it working properly.

SCEEPWARE, 2025-06-08:
For anyone having this issue, changing Windows theme color seems to fix it (it sounds like a bad joke but it isn't, didn't work when my accent color was grey but now that it's piss yellow it's all good).

Suchiman, 2025-07-25:
Likely fixed by upgrading WPF.UI to 3.0+, which adds exception handling to the `UnsafeNativeMethods.GetDwmColor` method